Ulnooweg/GroBot #11: Version display on LCD

TNarakol-UEC, 2024-10-22:
Add ability to check software version on LCD display to easily check if it needs updating

TNarakol-UEC, 2024-11-14:
@SeyitTumturk-UEC  I'm removing myself from this assignment now as it's only the LCD part remaining.

TNarakol-UEC, 2024-11-22:
@SeyitTumturk-UEC  sorry I changed how the structure work. The software version number is now the file "softver" in /code. This is accessible thru readcsv_softver() defined in config.py and returns a list that can be accessed as list[row][columns].

Don't use normal readcsv or writecsv for software version. Right now that is reserved for firmware changes only.

TNarakol-UEC, 2024-11-22:
@SeyitTumturk-UEC all of these are preferably in a new submenu called system info. With separate buttons to display current software version and current firmware version

SeyitTumturk-UEC, 2024-11-25:
Added under system_info() - in main menu.

closed 2024-11-25 by SeyitTumturk-UEC