Ulnooweg/GroBot #52: PiCamera takes pictures but doesn't return to menu on display

KMorris-UEC, 2025-02-27:
**Description**
After selecting "Take Picture Now" in the Manual Controls menu, the LCD screen becomes stuck on either a status message or a blank screen. After attempting to leave the blank screen, jumbled characters appear.

**Reproduction Steps**
Turn on the Grobot, select "Manual Control", then select "Take Picture Now",

**Expected Behavior**
The PiCamera should take a picture, then display whether it was successful. Then it should return to the menu.

**Actual Behaviour**
After selecting "Take Picture Now", it becomes stuck on a status message or blank screen.

**Screenshots/Logs/Supporting Media**

https://github.com/user-attachments/assets/44cc65c4-f5eb-4eab-a123-0950e643a232

[logfile.log](https://github.com/user-attachments/files/19010423/logfile.log)

TNarakol-UEC, 2025-02-27:
Possible solution - switch off multithreading in LCDDisp func for calling manual picam_capture(). This seems to fix it so will be extended to alll manual controls and edit settings function and tested. (Move everything off of threading)

TNarakol-UEC, 2025-03-25:
Fixed with picamera bug.